dwelch67/raspberrypi #12: How to generate kernel.img files?

surya2891, 2015-04-23:


surya2891, 2015-04-23:
i was able to pull it off, rather than using the cross compilers, i used the native one on my board and that saved my life :)
That was one of my worst nightmares yesterday, i should say!!

Thanks
Surya.

CJSoldier, 2015-12-24:
Just install minGW and yagarto on windows. After that check your environment whether it has string like this C:\yagarto-20121222\bin;C:\MinGW\bin;
If not,add it.Then open cmd window, input  cd /d blinkerxxx   enter input mingw32-make enter, done.
And you can also rename mingw32-make.exe to make.exe under C:\MinGW\bin, then just input make can also generate img file.

dwelch67, 2015-12-24:
I think I have a pitopi directory where it should tell you to remove the arm-none-stuff and just use gcc and as and ld.  you might try these pre-built cross compilers for linux or windows or mac.  https://launchpad.net/gcc-arm-embedded

cirosantilli, 2016-10-08:
On Ubuntu 16.04 simply:

    sudo apt-get install binutils-arm-none-eabi gcc-arm-none-eabi