dwelch67/raspberrypi #16: uart01.c - Wiring to AUX_IER_REG doesn't do anything?

nkolban, 2015-12-21:
Looking at the uart01.c sample, I see we write 0 to AUX_MU_IER_REG.   When I compare that to the BCM2835 spec, it appears that doesn't do anything or have any meaning.   Is this true?

dwelch67, 2015-12-21:
https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=72260 has a good list of resources  http://elinux.org/BCM2835_datasheet_errata shows mistakes in the broadcom document. see the p12 errata.  So the write is there to disable interrupts.  Why do I write that register twice?  Good question, I dont know maybe moved where I wrote it and didnt delete the prior line.

nkolban, 2015-12-21:
Ahha ... thank you sir ... going throughthe errata sheet now.... awesome response speed ... ty.

closed 2015-12-21 by nkolban

CJSoldier, 2015-12-22:
Hi,@David Welch.I followed bliner01 example,but I can't blink my led. I use Raspberry PI 2 Model B.
Should I modify anything?

CJSoldier, 2015-12-22:
Hi,@dwelch67. I sent you an e-mail,I guess you didn't receive it.
I followed bliker01 example,but I can't blink my led. 
I've already format my tf card as fat32 and I copied bootcode.bin and start.elf and kernel.img(blinker01.bin).
I use Raspberry PI 2 Model B. I tried many times,it didn't work.
Should I modify anything?

dwelch67, 2015-12-22:
yeah, try the beeplus example instead.  the led moved from gpio16 to gpio47 at some point (B+) and I have not gone and re-worked everything to handle the two peripheral address spaces, and other differences.

dwelch67, 2015-12-22:
the raspberry pi 2 also has a different address space so, hmmm bootloader07 has a blinker example possibly always builds both or you have to choose how to build so it builds the raspberry pi 2 blinker.  so there are three major variations, original processor/address space with gpio16, original with gpio47, and new processor/address space with gpio47.

dwelch67, 2015-12-22:
if nothing else I need to go back through and update the readme so say what combination they support.  if not go through and somehow come up with a way to solve all the combinations.