dwelch67/raspberrypi #17: I almost give up raspberry pi

CJSoldier, 2015-12-24:
I've learned David Welch's tutorial before buying raspberry pi2.
I'v been stucked in blinking the ACT LED in the past a few days.I thought I could succeed.
I was annoyed When I found that the GPIO base address in RPI2 was not 0x20200000 any more.
Today David updated his code which lift me up thought he is late :).
Hi,David, I read a tutorial from Brian who said that he knew RPI2's GPIO address from u-boot source code. How do you know that? The PI foundation doesn't supply enough datasheet which make worried about how far can we go with RPI's bare metal programming.

dwelch67, 2015-12-24:
There is nothing forcing you to do bare metal on the raspberry pi.  It is very rare that you have all the documentation you need, accurate documentation, that is a big part of bare metal not just wading through stuff, but filling in the gaps.  Even when you work where they make the boards (I dont work for the raspberry pi folks but we make boards where I work) the documentation is not always complete or accurate, engineers, humans, tend to do the work first then document later if ever.   

So we rely on what they do give us, we find stuff out through wading through forums, asking employees directly in the forums or looking through source code that they have provided.  Or the standard fall back just hack your way through it if nothing else try every gpio pin, maybe you get lucky.

It bothers me that Broadcom is not consistent they give out some amount of info, when the thing isnt popular yet or even a success (had the taste of the OLPC project before the boards shipped).  When the GPU was reverse engineered they actually documented that, but come the B+, A+ and pi2, little to nothing.  yet we figured out where the leds were didnt we, and even in my code had you looked around you would have found that information.  definitely some simple google searches would have found it.

So why use the rasberry pi for bare metal:

1) other than physical or electrical damage as far as we know they are not brickable
2) they are not brickable
3) they are not brickable
4) large, strong developer community
5) inexpensive
6) available

and on and on...I imagine there is stuff out there but not sure that before they covered their initial public orders on the first big run, there was already a developer community with a community drive errata sheet for the documentation.  I personally have not witnessed that before.  The from broadcom docs suck but with the community and large quantity of examples from many folks, this is still a very good bare metal platform (well it could use more gpio)...

I think the C.H.I.P is going to be a contender (notice the raspberry pi foundation felt they had to react to a kickstarter that hasnt proven itself yet, most dont), but it also has not great docs, but is quite likely brickable for the average user no doubt more brickable than the raspberry pi.  you cant remove the boot media without a rework station.

closed 2015-12-24 by dwelch67

dwelch67, 2015-12-24:
If it is guarantees you are after perhaps look at running programs on an open source sim like qemu.  100% documented in the source code for the simulator, no guessing required, nothing missing.

CJSoldier, 2015-12-25:
Thanks, David.I just want to understand computer in depth by R PI.
The PI reminds me of the time when I learned Principle of Micro-computer,I was interested in address.
I think it's how computer works.Sorry to say that I didn't understand it clearly.I know that the R PI foundation want to teach children program, I thought maybe someday I could tell children how computer works provided that I knew it better.Perhaps they won't understand, but I will try.
I remember that when I was a child, the only thing I did was playing with mud.