dwelch67/raspberrypi #18: Any chance you could team up with me-no-dev

KeithSloan, 2016-02-08:
I would like to see the ability to load Arduino sketches into the Raspberry Pi without an operating system. It looks like you have done a lot of the work of creating standalone bootloaders and me-no-dev has experience of interfacing with the Arduino IDE see https://github.com/me-no-dev/RasPiArduino Any chance you could team up.

C-Elegans, 2016-02-08:
The raspberry pi and the arduino are completely different computers running completely different architectures, so directly porting arduino code would be a no-go. Also, assuming I could integrate the arm compiler with the arduino IDE, most arduino libraries would not work. If you want to get started programming an rpi, I recommend python and rpi.GPIO.

KeithSloan, 2016-02-08:
>
> The raspberry pi and the arduino are completely different computers 
> running completely different architectures, so directly porting 
> arduino code would be a no-go.
Well an Arduino and a PIC32 chpkit Pi are two completely different 
architectures yet its possible to run sketches written for Ardunio on 
the chipKit Pi
See MPIDE http://chipkit.net/tag/mpide/. There is also 
http://www.uecide.org/

> Also, assuming I could integrate the arm compiler with the arduino 
> IDE, most arduino libraries would not work. If you want to get started 
> programming an rpi, I recommend python and rpi.GPIO.
As to arduino libraries see https://github.com/me-no-dev/RasPiArduino