dwelch67/raspberrypi #19: aarch64 gone?

BigEd, 2016-04-04:
Did you have some problem with the aarch64 examples for the Pi 3? Maybe I'm looking in the wrong place?

scotws, 2016-04-04:
I noticed that too, I'd love to have some Aarch64 examples for the RPi.

dwelch67, 2016-04-06:
sorry everyone they moved under boards/pi3  I am/was trying to deal with N number of boards with anywhere from no differences to many differences between them and up to 4 year old stuff written at the time there was only one board....so still have to work that out...like the other pi1 dirs (blinker01, etc) I should have left aarch64 with a readme...

dwelch67, 2016-04-06:
okay, I dont have much, dont necessarily need much as the C language part doesnt really change you just need a different compiler.  the bootstrap assembly changed yes to use the new instruction set.

The leds are behind an i2c gpio expander which I need to find/figure out, you can mailbox your way through the gpu to get there (see the raspberry pi bare metal forum) but I would like to someday figure that out.  And of course mmu/interrupts at some point in time...

other than that though it is still a case of once you get into C the instruction set doesnt matter so much.

BigEd, 2016-04-06:
Many thanks! (It's a pity in a way that the Pi platform is not a single standard configuration - but on the other hand it is in some ways improving.)

closed 2016-04-06 by BigEd

dwelch67, 2016-04-06:
we would still be driving the original model-T or a wright flyer if we
didnt have improvements.   why move the leds every Nth board?  gpio sure,
but they didnt add lots more stuff so they didnt need those pins.

I am glad the chips appear to for the most part have the arm core carved
out and a new one put in rather than a big do-over each time.  eventually
that will happen too, they will design around a different chip and maybe we
wont get info on it and have to reverse engineer linux sources...