dwelch67/raspberrypi #22: Running examples using qemu 2.6?

pykello, 2016-06-25:
Qemu 2.6 added support for raspi2. But I couldn't get the uart01 example run correctly using raspi2.

The steps I took was:
* Edit ```uart01/memmap``` and change origin to ```0x10000``` (which is the address qemu starts executing),
* make
* ```/usr/local/bin/qemu-system-arm -machine raspi2 -m 512M -nographic -gdb tcp::26000 -S -kernel uart01.bin```

Then start ```arm-none-eabi-gdb``` and run following commands:
``` 
target remote localhost:26000
symbol-file ./uart01.elf
```

Then, when I start the program, it seems that the inner while loop never breaks and I don't see any output.

I am wondering if I'm doing any steps incorrectly.

dwelch67, 2016-06-26:
qemu-system-arm -machine raspi2 -m 128M -nographic -kernel uart01.bin
qemu-system-arm: -machine raspi2: Unsupported machine type

Not sure which uart01 example, but if this is for a raspberry pi 2 then 
you need the peripherals to be based at 0x3F000000 not 0x20000000 that 
might help...

#define GPFSEL1 0x3F200004
#define GPSET0  0x3F20001C
#define GPCLR0  0x3F200028
#define GPPUD       0x3F200094
#define GPPUDCLK0   0x3F200098

#define AUX_ENABLES     0x3F215004
#define AUX_MU_IO_REG   0x3F215040
#define AUX_MU_IER_REG  0x3F215044
#define AUX_MU_IIR_REG  0x3F215048
#define AUX_MU_LCR_REG  0x3F21504C
#define AUX_MU_MCR_REG  0x3F215050
#define AUX_MU_LSR_REG  0x3F215054
#define AUX_MU_MSR_REG  0x3F215058
#define AUX_MU_SCRATCH  0x3F21505C
#define AUX_MU_CNTL_REG 0x3F215060
#define AUX_MU_STAT_REG 0x3F215064
#define AUX_MU_BAUD_REG 0x3F215068

without the gdb stuff it should just print to the console, at least in 
other qemu runs I can use ctrl-a then x to exit out.

DawidPi, 2017-08-11:
Hello,

I am also trying to use qemu. Uart01 will not work, as for me it seems, that simply only uart0 is redirected into console (maybe qemu does not yet support uart1). Go for uartx01 example It should work fine with steps you provided + changing peripherals offsets as David mentioned.

cirosantilli, 2018-06-19:
This worked well on QEMU rpi3:

- https://github.com/bztsrc/raspi3-tutorial 
- https://raspberrypi.stackexchange.com/questions/34733/how-to-do-qemu-emulation-for-bare-metal-raspberry-pi-images/85135#85135