dwelch67/raspberrypi #23: More detailed bootloader example steps

beriberikix, 2016-08-22:
I've tried a bunch of different combinations of upload to upload the src using the bootloader but without any luck. Would it be possible to share the exact commands/steps you use? 

For example, I am using `bootloader07` from `boards/pizero.` I got the expected serial output rather easily. I then modify an output string and try to upload. I am unable to successfully upload the .hex using either minicom or CoolTerm (I'm on a mac.)

Any guidance would be much appreciated!

dwelch67, 2016-08-23:
So with minicom if you are seeing the IHEX thing you are a long way 
there.  I assume no config.txt file is being used.

I use ctrl-a then press s and it gives you a text menu zmodem, ymodem, 
xmodem, kermit and ascii are what mine show.

I either up arrow once or down arrow a few to ascii.  then hit enter two 
times and it gives a  No file selected - enter filename:  Enter the 
path/filename to the .hex file.  My experience is that if you launch 
minicom in the same directory where you keep the .hex file you dont need 
to supply a path, their menu is clunky but functions if you want to 
navigate using the goto at the bottom and such.

Then after it has downloaded the file, press g and that causes the 
bootloader to jump to 0x8000 to run the program.

         if((ra=='g')||(ra=='G'))
         {
             uart_send(0x0D);
             uart_send('-');
             uart_send('-');
             uart_send(0x0D);
             uart_send(0x0A);
             uart_send(0x0A);
             BRANCHTO(0x8000);
             state=0;
             break;
         }

The program has to be built for 0x8000, see the blinker examples, build 
and try one of those.

David

beriberikix, 2016-08-23:
Thanks for the reply! I put together a short clip of what I'm seeing [here](https://youtu.be/ZJw4Yl4aDLQ).

In the video I try to show that serial is clearly working. I updated the string the gets outputted before I recompile. When I try to upload using `ascii`, it seems to fail as you can see from the output.

Any obvious things I'm doing wrong? Any suggestions?

beriberikix, 2016-08-23:
Update: using CoolTerm to send the file worked! Pressing `g` does the trick. Tried it with a few of the blinkers and uart examples. Not sure why the same doesn't work with minicom but I'll ignore that for now.

In case anyone else doesn't realize the obvious like me, the bootloader loads the file in memory and not onto the SD card. If you want to save something to load after restart you'll still have to right the kernel.img to the card. FYI!

closed 2016-08-23 by beriberikix

dwelch67, 2016-08-24:
Okay from the video you already have bootloader07 running you dont 
need/want to re-load bootloader from the bootloader, it should in theory 
stomp on itself.  you use the bootloder to load and run other programs 
like blinker01, etc.  any of the other programs not the bootloader.

David