dwelch67/raspberrypi #24: How to run the Raspberry Pi 2 blinker example?

cirosantilli, 2016-10-08:
What I've done so far:

- `sudo apt-get install gcc-arm-none-eabi binutils-arm-none-eabi`
- `make` inside `boards/pi2/bootloader07` and `boards/pi2/HYP/blinker01`. Seems to work, I get `kernel7.img`, `bootloader07.elf`, `blinker01.bin` and `blinker01.elf`
- format SD card with SD disk, create a single fat partition, copy the entire `bootloader07/*` and `blinker01/*` to root
- copy `bootcode.bin` and `start.elf` to the root partition as well from: https://github.com/raspberrypi/firmware/tree/master/boot

Now what do I have to do? I plug the SD card into Pi, and power it up, but nothing happens (both leds stay on all the time, as if I had no SD card).

I recommend that you add some very direct step by step bash commands at the some of one of the README, otherwise hard to make work. :-)

Ubuntu 16.04 host.

Thanks for awesome repo!

dwelch67, 2016-10-08:
you have a uart hooked up?  if not then the bootloader doesnt do you any 
good, try one of the blinkers by itself.

no reason to copy the bootloader directories, you only need the binary 
file and copy it to kernel7.img.  bootcode.bin, start.elf and 
kernel7.img are the only files you need. maybe config.txt if you want to 
change how it boots, but the subdirs in the repo are config.txt specific.

cirosantilli, 2016-10-09:
UART works on Raspbian through `/dev/ttyUSB0`, but I have no /dev/ttyUSB` with this image.

How to try a blinker by itself?

I had understood that only the img bin and elf were needed, only copied dir contents to be sure :-)

dwelch67, 2016-10-09:
From my readme file.

1) boots off of an on chip rom of some sort
2) reads the sd card and looks for additional gpu specific boot files
bootcode.bin and start.elf in the root dir of the first partition
(fat32 formatted, loader.bin no longer used/required)
3) in the same dir it looks for config.txt which you can do things like
change the arm speed, or change the address where to load kernel.img,
and many others
4) it reads kernel.img the arm boot binary file and copies it to memory
5) releases reset on the arm such that it runs from the address where
the kernel.img data was written

dwelch67, 2016-10-09:
this is bare metal not linux on the pi, you have no usb on the pi unless 
you write a bare metal usb driver (there have been a couple of folks to 
succeed at some level).  there is a ton of text related to getting a 
3.3v level uart that you connect to your host (an ftdi breakout or cable 
for example) with which you can connect to the uart on the pi and use a 
bootloader.  Without that you are pretty much done after you blink the led.

kernel.img or kernel7.img is the specific name of the file that contains 
the memory image copied to the arm and then booted. the four files 
listed are the only files that are used to boot the gpu and arm, one of 
them is optional config.txt.

David

cirosantilli, 2016-10-09:
I know this is bare metal, only mentioned Linux as way to ensure uart is connected correctly.

ftdi: OK will look into that, don't know what it is

kernel.img or kernel7.img: I understand, I just can't couldn't find the one that makes the blinker work, only the bootloader one which I couldn't use

dwelch67, 2016-10-09:
copy the binary boards/pi2/HYP/blinker01/blinker01.bin to kernel7.img on 
the card and power cycle.

cirosantilli, 2016-10-14:
:+1: 

And the partition type id must be `C`. Arg, 2 hours of my life went by. :-)

closed 2016-10-14 by cirosantilli