dwelch67/raspberrypi #25: a new way to avoid remove/insert SD card again and again

zhangboyang, 2016-10-28:
Recent firmware support network booting on pi2 (but not pi1), so we can setup a dhcp/tftp server, let the pi2 fetch kernel7.img from our server and then boot it.
I think it is easier to setup and do not require some special hardware.

this is my configuration:
[image]

screenshot:
[image]

Piping, 2017-10-05:
Hi, can you talk a bit more on how to setup the Ethernet and IP for the client?

So in bare metal environment what is required steps to achieve this?

KNNSpeed, 2017-10-05:
Why not go all the way and just SSH in instead of UART? The binary would be a bit big, but then you'd be able to do everything with 2 wires and have the flexibility to transfer large files.

Piping, 2017-10-05:
@KNNSpeed Hi, this is bare metal programming and SSH is just not there.

KNNSpeed, 2017-10-05:
@Piping I know, I'm saying it could be implemented at the cost of a rather large binary (and yes, it can be done)

Piping, 2017-10-05:
@KNNSpeed this is a problem like chicken and eggs. Anyway, I am just looking for Ethernet solution relies on the bootcode.bin that is provided by the pi. 

UART + boot loader to load program solution should be fine too, although I cannot get the code in this repo to work with pi3 B +.

KNNSpeed, 2017-10-05:
I have working uart bootloader code somewhere. Let me see if I can find it.

KNNSpeed, 2017-10-05:
Here you go. I haven't used this in a while, but I was able to transfer files over UART and press "g" to run them, no SD card swapping needed. Included is a little text files with the command line syntax to transfer stuff over terminal (for Mac, but linux isn't all that different).

[UART_Raspi3_Bootloader.zip](https://github.com/dwelch67/raspberrypi/files/1360867/UART_Raspi3_Bootloader.zip)

Piping, 2017-10-05:
@KNNSpeed  This is great! I appreciate that!!!

Piping, 2017-10-05:
@KNNSpeed  so correct me if I am wrong, dwelch67's bootloader load the code at 0x8000 position right? the bootloader is self resides in the 0x0?

KNNSpeed, 2017-10-05:
I... can't remember. It's been like a year and I didn't write it down... Hopefully I used the 64-bit stuff. You could send a program that just prints its memory address and find out (I'd like to know, too).

KNNSpeed, 2017-10-05:
Well, I also found these. Bootloader07 from aarch64:
[bootloader07_64bit.zip](https://github.com/dwelch67/raspberrypi/files/1360969/bootloader07_64bit.zip)

Bootloader07 from the root of this repo:
[bootloader07.zip](https://github.com/dwelch67/raspberrypi/files/1360972/bootloader07.zip)

I don't remember if these actually work, though. Like I said it's been a while...

bluecereal, 2017-10-06:
@Piping check out the memmaps in some of the examples.  like [this](https://github.com/dwelch67/raspberrypi/blob/master/bootloader01/memmap) and [this](https://github.com/dwelch67/raspberrypi/blob/master/blinker01/memmap).

bluecereal, 2017-10-06:
also [this](https://github.com/dwelch67/raspberrypi/blob/master/blinker01/vectors.s#L7).  basically yes dwelch67's bootloader loads the code at 0x8000.

KNNSpeed, 2017-10-06:
@Piping Looks like the simple answer to your question is yes.

Thanks @bluecereal!

Piping, 2017-10-07:
@bluecereal @KNNSpeed  Thanks for your reply.

I am testing with the aarch64 code inside the board/pi3/aarch64. 

I set up my config file with `kernel_old=1` and other settings in @dwelch67 's repo.

The bootloader07 can print 12345678 and ihex string as expected. Although the run doesnt load with `cu` program, while minicom is just too hard to learn to use.

I use the following command if anyone interest.

`sudo cu -s 115200 -l /dev/cu.usbserial`
and use type `~>` to send a file. But it only prints 1 R R R R timeout error. I am not sure what is going here.  I used serial cable from uart to ttl with a chip called P2303.

I also used the toolchains from linaro, `aarch64-elf-gcc` for example.

timanu90, 2017-10-07:
For terminal, I like to use GTKTERM is very simple. To send a file just go file->send raw file select your gcc generated file in format ihex and done. Works great give it a try.

KNNSpeed, 2017-10-08:
I provided the exact command line to use for picocom, which is a pretty simple program in the above zip file. I agree minicom is pretty tough to use. If you do what I wrote in the text file, you just specify the file path (you can drag 'n' drop depending on your OS), and it sends the file. I think the only difference between Mac using MacPorts and linux is that on linux /dev/tty.usberserial is actually  /dev/tty[somethingelse]

Piping, 2017-10-08:
@KNNSpeed Thanks again. I install the picocom using brew. And it works great!! 
@timanu90 Hi, I am on a mac computer and picocom is just easier to get. Thanks anyway!

Piping, 2017-10-09:
@KNNSpeed nvm, when I turn picocom local echo on I can see the characters I typed in appeard twice.

KNNSpeed, 2017-10-09:
Is the Pi echoing back anything? That might be why.
As I said I don't have a Uart cable or bare pi handy, so I can't do much in the way of checking. :/

Piping, 2017-10-09:
@KNNSpeed The code does work as echo back what I am typing. The picocom itself does not echo the character i typed. That's why I was confused.

zhangboyang, 2017-10-13:
@Piping What I did was copy (and only copy) bootloader to SD card, set up a DHCP and a TFTP server, put all boot files to TFTP root directory. Then when rpi2 starts up, it will try network booting.
However, I did it about a year ago and I can't recall the details. But it seems you have succefully used @KNNSpeed 's bootloader to solve the same problem. So the details is no longer useful. :)